TengerTechnologies/Smoothificator #1: Outer perimeter only 2 layes tall Orca 2.2.0

PorpoiseHork, 2025-01-25:
Tried this with Orca 2.2,0 and Python 3.13

Gcode only shows two layers on external perimeter. 

Tried following variants of script. 

"C:\Users\****\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\***\Documents\SV07 Klipper Additions\Smoothificator.py" -o 0.1; 
Resulted in Error Code: 2 when exporting gcode. 

"C:\Users\Bretf\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\Bretf\Documents\SV07 Klipper Additions\Smoothificator.py" -outerLayerHeight 0.1;

Successful export but when reviewed outer perimeter has just two layers. 

[SV07 Foot_PLA_Test 0.2_53m36s.zip](https://github.com/user-attachments/files/18547673/SV07.Foot_PLA_Test.0.2_53m36s.zip)

TengerTechnologies, 2025-01-25:
I think I know what the issue is. I can't fix it right now because I don't have the time to test wether it breaks something or not. But you should be able to easily fix it by changing the line 85 from if line.startswith("G1 Z"): to         
if line.startswith("G1 Z") or line.startswith(";Z"):

For your specific case this should work. But please don't close the issue until really fixed

PorpoiseHork, 2025-01-25:
Unfortunately that did not fix it and other models like the one attached now has large gaps on the model.  I tested different start gcode profiles as well as older versions of Python just to see if that made any difference but kept getting Error 2's with Orca.  I included a non script version as well as the edited script that you can check as time allows.  Thanks in advance. 

[SV07 Script.zip](https://github.com/user-attachments/files/18548234/SV07.Script.zip)

PorpoiseHork, 2025-02-02:
Tried the edits and while it got better it's still not working correctly. It's just randomly switching between the set layer of .1 and the internal of .20.  Some models like  the script doesn't even seem to be applied to the outer layer at all and just outputs it with matching interior layers. Others come out like this and are the same for internal layers of .24 and .28 Looking at the log the only thing I see that's out of place is where it says "Layer 1 detected at Z=25.000"  I suspect it's flagging that from my start gcode where I have it set to do the following after completing the adaptive mesh.

G1 Z25 F240
G1 X2 Y10 F3000
M104 S{first_layer_temperature[0]} ; set final nozzle temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize
**_G1 Z0.28 F240_**
G92 E0
G1 Y240 E30 F1500 ; prime the nozzle
G1 X2.8 F5000
G92 E0
G1 Y10 E30 F1200 ; prime the nozzle
G92 E0

[Smoothificator Script.zip](https://github.com/user-attachments/files/18632377/Smoothificator.Script.zip)

. 

[image]

sihawken, 2025-02-13:
Give my fork a try and see if it works for you in Orcaslicer.  Works well on my system.

https://github.com/sihawken/Smoothificator/blob/main/Smoothificator_Adaptive.py

PorpoiseHork, 2025-02-15:
@sihawken I tried it and it does seem to be working. It just doesn't like it when supports or scarf joints are used. Other than that it does appear to work correctly.

sihawken, 2025-02-16:
I already knew that scarf joints don't work, I have a few thoughts on how to solve that problem though. 

I haven't noticed that it doesn't like supports,  but honestly I am not surprised, I avoid supports like the plague. So I wouldn't have noticed.